wireviz/WireViz #434: Cut list

lutorm, 2024-11-30:
Beyond the BOM, the second thing that's needed is a cut list that then tells me how the 80m of wire the BOM tells me I need to buy has to get cut up into the various cables... Although I guess right now this is hard to do given that you can't route a wire continuously between different cables in wireviz, so it would make it look like you needed to cut it where you didn't actually.

kvid, 2024-12-23:
@lutorm wrote:
> Beyond the BOM, the second thing that's needed is a cut list that then tells me how the 80m of wire the BOM tells me I need to buy has to get cut up into the various cables... Although I guess right [...]

I'm sorry this issue has been unanswered for several weeks.

I agree that a cut list would be a nice feature. Please also suggest an output format.

You are right, it's hard to do with the currently latest version. First, we need to add a feature to enable wires to continue uncut through several bundles. See my https://github.com/wireviz/WireViz/issues/270#issuecomment-2123078659 where I comment a suggested syntax for such a feature - using what I call a virtual splice that is described as a work-around in my https://github.com/wireviz/WireViz/issues/270#issuecomment-1029535877, and preferably also combining it with my suggested feature #273.

lutorm, 2024-12-26:
I think the format for the cutlist could most simply be just like the BOM, only instead of concatenating all the designators into one line, they would be on separate lines with their individual lengths.

kvid, 2024-12-26:
@lutorm wrote:
> I think the format for the cutlist could most simply be just like the BOM, only instead of concatenating all the designators into one line, they would be on separate lines with their individual lengths.

To allow outputting both a BOM and a cut list from the same run, we would need separate output format specifiers and separate filenames, though.

Another functionality that might be useful for cut lists, is to optionally allow flagging any connector/pin entry in connection lists as a wire continuation (similar to virtual splices mentioned above) to support also mid-wire connections, like e.g. IDC ribbon cable connectors, IDC 3-way splices, mid-wire soldered daisy-chained wire loops, etc.

lutorm, 2024-12-26:
In my mind, the cut list is strictly a thing you go through cutting wires to the correct lengths. What you're suggesting sounds more like a by-wire build instruction, like "Wire X consists of a terminal pin, conductor A, a splice, conductors B, a socket, conductor C, and a ring terminal"? I agree this is also something that's needed to be able to make sense of a large harness, but it seems distinct from a cut list.

kvid, 2024-12-26:
I don't recognize my latest suggestion in your description above. My description was probaly not good enough, so I try again - using an example this time: When using WireViz to specify e.g. a 50 cm ribbon cable with one IDC connector at each end plus one at the middle, it might look like this:
```yaml
connectors:
  IDC connector:
    pincount: 10
cables:
  Ribbon cable:
    wirecount: 10
    length: 0.25 m
connections:
  -
    - IDC connector.Left [1-10]
    - Ribbon cable [1-10]
    - IDC connector.Middle [1-10]
    - Ribbon cable. [1-10]
    - IDC connector.Right [1-10]
```
This would result in two 25 cm entries in a cut list unless we can specify that the same cable continue uncut through the middle connector.