wireviz/WireViz #439: Add a way to add a loopback wire, add example of it

DeflateAwning, 2024-12-12:
I want to add a tiny loopback wire as shown in the following diagram (pink jagged line).

The rest of the diagram is below. 

This would be great to add as an example if it's currently possible; otherwise, it'd be a great feature to add! 

[image]

<details>

```yaml
connectors:
  CN1 (AC Input):
    type: JST B3P-VH on PCB, JST SVH-21T-P1.1 on wire, JST VHR housing
    subtype: female
    pinlabels: [Neutral, N/C, Line]
  CN2 (DC Output):
    type: JST B8P-VH on PCB, JST SVH-21T-P1.1 on wire, JST VHR housing
    subtype: female
    pinlabels: [+V, +V, +V, +V, -V, -V, -V, -V]
  CN3 (Power Good):
    type: JST B4B-XH on PCB, JST SXH-001T-P0.6 on wire, JST XHP housing
    subtype: female
    pinlabels: [PG, GND, Neg Sense, Pos Sense]
  CN901 (Standby):
    type: JST B-XH on PCB, JST SXH-001T on wire, JST XHP housing
    subtype: female
    pinlabels: [N_PS_ON, GND, 5VSB, GND]

  AC Input:
    subtype: female
    pinlabels: [Line, N/C, Neutral]

  Main Connector:
    subtype: female
    pinlabels: [V1, V1, V1, RTN, RTN, RTN, Neg Sense, Pos Sense, Remote PS ON]

cables:
  AC Wire:
    # gauge: 0.25 mm2
    # length: 0.2
    color_code: DIN
    wirecount: 2

  Main Wire:
    # gauge: 0.25 mm2
    # length: 0.2
    color_code: DIN
    wirecount: 9

  Pigtail Loopback:
    # gauge: 0.25 mm2
    # length: 0.2
    color_code: DIN
    wirecount: 1

connections:
  -
    - CN1 (AC Input): [3, 1]
    - AC Wire: [1, 2]
    - AC Input: [1, 3]

  -
    - CN2 (DC Output): [1-3, 5-7]
    - Main Wire: [1-6]
    - Main Connector: [1-6]

  -
    - CN3 (Power Good): [3, 4]
    - Main Wire: [7, 8]
    - Main Connector: [7, 8]

  -
    - CN901 (Standby): [1]
    - Main Wire: [9]
    - Main Connector: [9]
```
</details>

tobiasfalk, 2024-12-15:
What you described here in pink would not be a loop back wire since it goes from one connector to the other.

A way to short and loop wires inside one connector is worked on, https://github.com/wireviz/WireViz/pull/412

To do what you want, I would create its own wire.

kvid, 2024-12-16:
@tobiasfalk wrote:
> What you described here in pink would not be a loop back wire since it goes from one connector to the other.
> 
> A way to short and loop wires inside one connector is worked on, #412

What @tobiasfalk is trying to explain, is that this feature request cannot be solved with the existing optional `loops` attribute of a connector, because those loops are only supported between pins at the same connector, and PR #412 tries to improve the existing functionality, but is still limited to loops or shorts between pins at the same connector.

> 
> To do what you want, I would create its own wire/cable.

A quick and dirty work-around to obtain what @DeflateAwning asks for, is to append these connections:
```yaml
  -
    - CN2 (DC Output): [8]
    - Pigtail Loopback
  -
    - CN901 (Standby): [2]
    - Pigtail Loopback
```
However, the rendering is not very pretty. It looks almost like two wires are spliced:
[image]
This method also opens up for more than two connections for the same wire, which is normally not possible in real life without some kind of splicing or mid-wire connections. A future version of WireViz might therefore give a warning or error about this.

Another work-around requires more know-how about how WireViz generates the Graphviz input:
```yaml
# The connectors are unchanged
cables:
  # The other cables are unchanged
  Pigtail Loopback:
    colors: [PK, PK]
    show_wirecount: false
    show_wirenumbers: false
connections:
  # The other connections are unchanged
  -
    - CN2 (DC Output): [8]
    - Pigtail Loopback: [1]
  -
    - CN901 (Standby): [2]
    - Pigtail Loopback: [2]
tweak:
  append: |
    "Pigtail Loopback":w1:e -- "Pigtail Loopback":w2:e [label=" " color="#000000:#ff66cc:#000000"]
```
And it looks a bit better:
[image]
However, the BOM will show that "Pigtail Loopback" is a 2-wire cable unless specifying `category: bundle`, but if doing that and specifying a non-zero length, it will be doubled in the BOM because we use two wires looped together.

kvid, 2024-12-22:
1. A possible new feature might be to detect two connections to the same wire from the same side (e.g. the input of my first quick and dirty work-around above), and render that similarly to my second work-around above without doubling the wire in the BOM. If more than two connections are detected to the same wire (in total from either side), then either:
- Raise an exception that 3 or more connections to the same wire are not supported, or
- Maybe as a future extension: For each connection (except the first) on each side, add wires with a loop in the same bundle (or separate anonymous bundles), so the loops are daisy-chained.
2. Another possible new feature might be to expand the syntax to allow (optionally) flagging a wire entry in a connection as a U-turn, that creates a wire loop and makes the following entries to be right-to-left until the end or to the next U-turn that makes the following entries to be left-to-right again. This might be a more user friendly alternative, especially for anonymous wire loops. Such a flagging could e.g. be `<U>` in front of the wire entry.
3. As an extension to the latter possible new feature above, the same optional U-turn flagging could also be allowed for pin entries in a connection. This might be a more user friendly alternative for daisy-chaining (both daisy-chaining wire loops and daisy-chaining straight wires).