wireviz/WireViz #441: GitHub Actions fixed by #408 is still failing

kvid, 2025-01-15:
This accepted answer seems relevant: https://stackoverflow.com/questions/67743583/how-can-i-stop-github-actions-from-requiring-old-nodejs-builds

I find the old actions as required before merging into `dev` or `master`, and it seems I can remove them as required, but I don't find the new actions to replace them. What am I missing?

Update: I probably need to cherry-pick the squash commit from mege-in of this PR also into the target `dev` branch before correcting the branch protecting rules, but I must wait testing this until I have access to my PC again.

_Originally posted by @kvid in https://github.com/wireviz/WireViz/issues/408#issuecomment-2559808879_

kvid, 2025-01-15:
By cherry-picking the commit described above also into the `dev` branch, I finally was able to make the branch protecting rules using the new actions, and it stopped failing due to not finding the old actions anymore.

However, the new actions also fail, like this: https://github.com/wireviz/WireViz/actions/runs/12736147371

"The version '3.7' with architecture 'x64' was not found for Ubuntu 24.04." is the failing action, and the remaining operations were canceled.

It seems the "ubuntu-latest" specifyer started to use ubuntu-24.04 instead of ubuntu-22.04 recently, and ubuntu-24.04 only support Python 3.9, 3.10, 3.11, and 3.12, but has stopped supporting Python 3.7 and 3.8 - and hence those actions are failing. For more details, see https://github.com/actions/runner-images/issues/10636

If we still want actions that uses Python 3.7 and 3.8, then we must specify ubuntu-22.04 for those. Otherwise, we can just drop supporting Python 3.7 and 3.8 from future versions, but then we also need to update README and setup files.

amotl, 2025-01-15:
Yes, exactly. For supporting EOL versions of Python, you need to run it on `ubuntu-22.04` or even `ubuntu-20.04` when looking at Python 3.6 ;].

formatc1702, 2025-01-16:
I would have no issue with dropping support for EOL versions, so even if there is a fix by using old Ubuntu distros, it's good to be able to shake off old baggage, and we don't have the resources to keep worrying too much about backward compatibility, especially if it means not being able to use new features.

So from my side, absolute green light to just drop support for old versions.

If the issue persists, it's also OK to temporarily change the settings and to not require passed checks for merge to be allowed, as long as the tests pass with current Python versions.

If this is something only I can do, and you need it, feel free to ping me; mail is also fine.

kvid, 2025-03-14:
My commit e4fe099f8c7b86736aee7b4227cc794b6e8b36f0 into master in February seems to have resolved this issue.

PR #442 (2025-01-27): Fix GitHub Actions matrix to support the Python version range
- Fix #441 
- Use ubuntu-latest for the supported Python versions
    - Use ubuntu-22.04 only for Python 3.7-3.8
- Update setup.py with all tested python versions
- Required actions for the `dev` branch seems to work as intended with these new actions.
- Required actions for the `master` branch should later also work as intended by cherry-picking the first commit from `dev` to `master`.
  - The second commit updating `setup.py` should wait until just before the next release.
  - A rebase-merge is needed to keep these two commits sep [... cut]
closed 2025-03-14 by kvid