wireviz/WireViz #445: Image Types

kvid, 2025-02-22:
I noticed that the SVG image for a connector is not shown in the PNG output. I wonder if this is also observed by others. JPG, PNG, and GIF images in connectors seems to work fine, but a TIF image was rejected by Graphviz. It would be nice to create a test use case with all image types expected to work, and perhaps a separate issue for unexpected outputs for such a test use case.

_Originally posted by @kvid in https://github.com/wireviz/WireViz/issues/443#issuecomment-2645515636_

Preliminary test use case YAML input:
```yaml
connectors:
  png: &png
    pincount: 4
    image:
      src: resources/MolexKK254_4-pin.png
  jpg:
    <<: *png
    image:
      src: resources/MolexKK254_4-pin.jpg
  svg:
    <<: *png
    image:
      src: resources/MolexKK254_4-pin.svg
  gif:
    <<: *png
    image:
      src: resources/MolexKK254_4-pin.gif
#  tif:
#    <<: *png
#    image:
#      src: resources/MolexKK254_4-pin.tif
#  webp:
#    <<: *png
#    image:
#      src: resources/MolexKK254_4-pin.webp
cables:
  w:
    category: bundle
    colors: [YE]
    length: 1
connections:
  - - png: [1]
    - w.: [1]
    - jpg: [1]
    - w.: [1]
    - svg: [1]
    - w.: [1]
    - gif: [1]
#    - w.: [1]
#    - tif: [1]
#    - w.: [1]
#    - webp: [1]
```
Image files expected to be in a `resources` subfolder:
[image]
[image]
[image]
[image]

kvid, 2025-02-22:
My host OS:
```
> wsl -l -v && ver
  NAME            STATE           VERSION
* Ubuntu-20.04    Running         1
  Ubuntu          Running         1

Microsoft Windows [Version 10.0.19045.5371]
```
My local environment (used `pip3 install -e .` to install WireViz):
```
$ git status && wireviz -V && python -V && dot -V && lsb_release -d
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean

WireViz 0.4.1
Python 3.8.10
dot - graphviz version 2.43.0 (0)
Description:    Ubuntu 20.04.4 LTS
```
Running test use case:

```
$ wireviz issue445.yml

WireViz 0.4.1
Input file:   issue445.yml
Output file:  issue445.[html|png|svg|tsv]
Warning: No loadimage plugin for "svg:cairo"
```
Output:
[image] `issue445.png`

[image] `issue445.svg`

[image] `issue445.svg` if PR #443 is accepted

TheMDev, 2025-02-22:
Interestingly enough wireviz immediately crashes macOS when running your example file with:
```
WireViz 0.4.1
Input file:   wire.yml
Output file:  wire.[html|png|svg]
Traceback (most recent call last):
  File "/Users/michaelwagner/.local/pipx/venvs/wireviz/lib/python3.12/site-packages/graphviz/backend/execute.py", line 88, in run_check
    proc.check_returncode()
  File "/usr/local/Cellar/python@3.12/3.12.9/Frameworks/Python.framework/Versions/3.12/lib/python3.12/subprocess.py", line 504, in check_returncode
    raise CalledProcessError(self.returncode, self.args, self.stdout,
subprocess.CalledProcessError: Command '[PosixPath('dot'), '-Kdot', '-Tpng', '-O', 'wire']' died with <Signals.SIGABRT: 6>.
```
I'm going to look further into this issue later but it seems Graphviz on macOS doesn't like jpg and gif.
However, just using the png and svg you can see the scaling issue that does not appear to be present in yours. 

[image]
`issue445.png`

[image]
`issue445.svg`

kvid, 2025-02-22:
Thank's for responding super-quickly! Please edit your comment above to add the versions of your OS, Python, and Graphviz. The warning I got from Graphviz also indicates there might be some optional plugins that could make a difference, so I edited my comment above to use `dot -v` for also outputting plugin information.

TheMDev, 2025-02-24:
After running dot-v and comparing our outputs, I was confused by the gap in the versions. Graphviz 2 is some 10 years old at this point. It appears Debian and Ubuntu repos have decided not to update, and I was unable to find 3rd party repos that provide a recent version as well. The only choice for Debian and Ubuntu is to download or build from the source. Interestingly enough, Fedora has the same version of Graphviz as I have on macOS.

`macOS Sequoia 15`
`Python 3.12.7`
```
dot - graphviz version 12.2.1 (20241206.2353)
libdir = "/usr/local/Cellar/graphviz/12.2.1/lib/graphviz"
Activated plugin library: libgvplugin_dot_layout.6.dylib
Using layout: dot:dot_layout
Activated plugin library: libgvplugin_core.6.dylib
Using render: dot:core
Using device: dot:dot:core
The plugin configuration file:
	/usr/local/Cellar/graphviz/12.2.1/lib/graphviz/config6
		was successfully loaded.
    render	:  cairo dot dot_json fig gd json json0 map pic pov ps quartz svg svg_inline tk vrml xdot xdot_json
    layout	:  circo dot fdp neato nop nop1 nop2 osage patchwork sfdp twopi
    textlayout	:  textlayout
    device	:  bmp canon cgimage cmap cmapx cmapx_np dot dot_json eps exr fig gd gd2 gif gv icns ico imap imap_np ismap jp2 jpe jpeg jpg json json0 kitty kittyz pct pdf pic pict plain plain-ext png pov ps ps2 psd sgi svg svg_inline svgz tga tif tiff tk vrml vt vt-24bit vt-4up vt-6up vt-8up wbmp webp xdot xdot1.2 xdot1.4 xdot_json
    loadimage	:  (lib) bmp eps gd gd2 gif jpe jpeg jpg pdf png ps svg webp xbm
```