wireviz/WireViz #458: Request: Connecting Shield to a Wire

georgina-bl, 2025-04-11:
Hello, I'm really liking Wireviz and it has been very useful in creating wire diagrams. All of the many features have been very useful thus-far, and I have been able to make all of my wire diagrams with no issues.

I was wondering if you are still updating the software, and if so; would it be possible to implement a new feature whereby you can connect a shield to a wire. I was wondering because for some of the wires we make, we trim the shield back and roll it into a small tail, then solder another wire to the shield tail, and put heatshrink over the connection. The wire is then connected to a solder tag. 

The existing feature is great as I can connect the shield to a connector, but I was wondering if it was possible to connect the shield to a wire, then a connector.

kvid, 2025-05-27:
I'm sorry that I've been offline and not available for contributing for many weeks.

@georgina-bl - Thank you for sharing your use case. Connecting a wire directly to another wire is not currently supported, and what you describe is soldering a cable shield to a wire. Why not represent the solder junction by a simple splice connector, like in [example 14](https://github.com/wireviz/WireViz/tree/master/examples#example-14)? If you really don't want the juction to look like a connector, you might achieve this by using what I call a virtual splice work-around described in https://github.com/wireviz/WireViz/issues/270#issuecomment-1028188154 and improved in https://github.com/wireviz/WireViz/issues/270#issuecomment-1029535877.

georgina-bl, 2025-05-29:
Hello, thanks so much for your response! I have not used the splice connector before but from looking at the comment you linked I think that will work perfectly for what we need. I will have a try at using the virtual splice as I really like the look of that one. Thanks again for your help!

kvid, 2025-06-12:
Be aware that the goal of my virtual splice work-around is to make an invisible splice between wire sections that have to be defined separately due to WireViz limitations, but that are one uncut continuous wire in real life. However, my initial attempts showed a tiny splice box at the junction point, and that might be closer to something useful for your use case?