wireviz/WireViz #460: Add a way to add a title to the images

DeflateAwning, 2025-04-26:
Either a titleblock or a top-of-page title would be good.

If this does exist, it should be added to more examples.

snelweg, 2025-04-27:
You can use the `template` function for that? The title block only shows up in the .html file, but you could export that to a pdf and have the titleblock integrated.

``` yaml
metadata: &metadata 
  title: Power supplies
  pn: PSU

  authors:
    Created:
      name: xxx
      date: 2025-03-11
    Approved:
      name: xxx
      date: 2025-03-11

  revisions:
    A:
      name: Ton
      date: 2025-03-12
      changelog: updates]

  template:
    name: din-6771    
    sheetsize: A4
```

https://github.com/wireviz/WireViz/tree/master/src/wireviz/templates

DeflateAwning, 2025-04-28:
Awesome, thanks! My request remains that adding a title to PNG exports would be very helpful!

kvid, 2025-06-13:
This issue seems a bit similar to #227 and #428.