wireviz/WireViz #468: Variants and varying lengths

BugInACode, 2025-06-20:
Hi,

I sometimes have the same harness that comes in varying lengths (for example one harness comes in 5 different lengths). 
Back when I was using WYSIWYG software we simply wrote `L mm` and had a little box with variants distinguishing the different lengths.

The harnesses usually had a suffix which determined which variant (such as a -1, -2, -3, etc) and in the table the numbers corresponded to the respective lengths. 

It would be great if there was a feature which could generate the relevant variants automatically and be able to specify this in the yaml file.

An example would be something like

```
cables:
    W1:
        type: coax
        length: &L # <-- here a variable is generated
        wirecount: 1
        colors: [BN]
        shield: true
        pn: RG179
        manufacturer: multicomp PRO
        additional_components:
        -
            type: heatshrink
            subtype: adhesive
            qty: &H # <-- here another variable is generated
            unit: mm
            qty_multiplier: terminations
```
Now that the length is assigned as a variable I'd like to be able to specify the different length later such as

```
variants:
    L: 
        [200, 400, 1000, 1600, 2400] mm # this automatically assigns the suffix -1 -> -5
        10: 3000 mm  # this method manually assigns the -10 to 3000 mm
    H: 
        1: 25 mm # this method manually assigns the -1 to 200m
        5: 50 mm # this method manually assigns the -5 to 500m
        10: 100 mm # this method manually assigns the -10 to 1000m
```
This should then generate a table on the output that ideally looks as such:

| Variant  | L | H |
| ------------- | ------------- | ------------- |
| 1  | 200 mm  | 25 mm  |
| 2  | 400 mm  | -  |
| 3  | 1000 mm  | -  |
| 4  | 1600 mm  | -  |
| 5  | 2400 mm  | 50 mm  |
| 10  | 3000 mm  | 100 mm  |

I'm open to improving this, especially how the variables are specified in the file, and also comments related to if information for a variant is left blank or whether it must be specified.

kvid, 2025-06-27:
Interesting idea. Have you considered a preprocessing script that generates the different variant values and calls WireViz for each variant?

If you use `*` instead of `&` where using a variable in the common harness YAML file, then the variables can be defined as YAML anchors in generated YAML files for each variant and prepend one generated file (with ` --prepend`) when calling WireViz.

Generated `variant-1.yaml`:
```yaml
variant anchors:
    Variant: &V 1
    Length: &L 200 mm
    Heat shrink: &H 25 mm
```
Common `harness.yaml`:
```yaml
cables:
    W1:
        type: coax
        length: *L # <-- here a variable is used
        wirecount: 1
        ...
```

BugInACode, 2025-06-30:
@kvid thanks for checking into this idea.

It is preferred that the one file/page is used as the final drawing should have the same name. For example HARN005-002A but when holding a physical harness, it should have a tag that says HARN005-002A-1 or HARN005-002A-2 depending on the length.

This means someone can pick up the harness, know which project it's from, the unique identifier, as well as revision so that they know which drawing it came from. But it also gives the person an instant ability to know which variant it is. We also use colour tags for different lengths so the harness may even omit the -1 and only be distinguished via the colour.

I like the idea of using variables for the name of the files as we plan to not append the yaml files with revision numbers but the outputs will use them. For example, HARN001-03.yaml will generate HARN001-03A.html (or PDF or whatever) but then if the yaml file is updated with a newer revision B, the yaml file should generate HARN001-03B.html without touching *A to maintain revision control.