wireviz/WireViz #476: Bug: Can't use JPG images

SiboVG, 2025-09-13:
Hi, first of all, thank you so much for this amazing tool, it's a life saver for my line of work!

Now, I do experience an issue when using images. I can use PNG images perfectly fine, but when trying to use a JPEG, I get this error:

```bash
WireViz 0.4.1
Prepend file: wireviz_connectors.yaml
Prepend file: wireviz_cables.yaml
Input file:   harnesses_wireviz_mnt_1.yaml
Output file:  harnesses_wireviz_mnt_1.[html|png|svg|tsv]
Traceback (most recent call last):
  File "/Users/sibo/.pyenv/versions/3.11.9/lib/python3.11/site-packages/graphviz/backend/execute.py", line 88, in run_check
    proc.check_returncode()
  File "/Users/sibo/.pyenv/versions/3.11.9/lib/python3.11/subprocess.py", line 502, in check_returncode
    raise CalledProcessError(self.returncode, self.args, self.stdout,
subprocess.CalledProcessError: Command '[PosixPath('dot'), '-Kdot', '-Tpng', '-O', 'harnesses_wireviz_mnt_1']' died with <Signals.SIGABRT: 6>.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/sibo/.pyenv/versions/3.11.9/bin/wireviz", line 7, in <module>
    sys.exit(wireviz())
             ^^^^^^^^^
  File "/Users/sibo/.pyenv/versions/3.11.9/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/sibo/.pyenv/versions/3.11.9/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/Users/sibo/.pyenv/versions/3.11.9/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/sibo/.pyenv/versions/3.11.9/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/sibo/.pyenv/versions/3.11.9/lib/python3.11/site-packages/wireviz/wv_cli.py", line 141, in wireviz
    wv.parse(
  File "/Users/sibo/.pyenv/versions/3.11.9/lib/python3.11/site-packages/wireviz/wireviz.py", line 385, in parse
    harness.output(filename=output_file, fmt=output_formats, view=False)
  File "/Users/sibo/.pyenv/versions/3.11.9/lib/python3.11/site-packages/wireviz/Harness.py", line 684, in output
    graph.render(filename=_filename, view=view, cleanup=cleanup)
  File "/Users/sibo/.pyenv/versions/3.11.9/lib/python3.11/site-packages/graphviz/_tools.py", line 171, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/sibo/.pyenv/versions/3.11.9/lib/python3.11/site-packages/graphviz/rendering.py", line 122, in render
    rendered = self._render(*args, **kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/sibo/.pyenv/versions/3.11.9/lib/python3.11/site-packages/graphviz/_tools.py", line 171, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/sibo/.pyenv/versions/3.11.9/lib/python3.11/site-packages/graphviz/backend/rendering.py", line 326, in render
    execute.run_check(cmd,
  File "/Users/sibo/.pyenv/versions/3.11.9/lib/python3.11/site-packages/graphviz/backend/execute.py", line 90, in run_check
    raise CalledProcessError(*e.args)
graphviz.backend.execute.CalledProcessError: Command '[PosixPath('dot'), '-Kdot', '-Tpng', '-O', 'harnesses_wireviz_mnt_1']' died with <Signals.SIGABRT: 6>. [stderr: b'']
```

The cable in question:

```yaml
W_B2B_1:
  type: Double-Row IDC, Double End
  wirecount: 8
  gauge: 28 AWG
  manufacturer: Samtec
  mpn: IDSD-04-D-09.00
  length: 0.2286
  colors: [BN, RD, OG, YE, GN, BU, VT, GY]
  bgcolor_title: '#BCAAA4'
  image:
    src: 'img/Samtec/IDSD-04-D.jpg'
    width: 150
```

I'm running the latest version of WireViz, and am on macOS 15.6.1. I've tried several JPEG images, they all experience the same issue.

tobiasfalk, 2025-09-13:
Hi, can you maybe try to only generate the HTML output?
Since it looks to me that it can not convert the Jpeg to png.

SiboVG, 2025-09-13:
I managed to resolve this by upgrading my graphviz installation.

closed 2025-09-13 by SiboVG